glikely/obs-ptz #158: How to build for windows?

markusd1984, 2023-04-29:
Either the files have been removed/changed or missing for the [windows build instructions](https://github.com/glikely/obs-ptz/tree/main#windows)?

> Run scripts\winbuild.cmd to build the plugin
> Run scripts\winrun32.cmd or scripts\winrun64.cmd to run OBS either the 32bit or 64bit version of the plugin

is missing in the scripts folder, as well as `scripts\winbuild-rel.cmd`

I tried running `windows.ps1` in PowerShell but had no success as it's looking for a "release" folder that doesn't exist.
Please help / advice, many thanks!

GeNTooGeek, 2023-08-21:
Possibly a little late to the game, but I found the build script in .github\scripts\Build-Windows.ps1.  The script accepts parameters for choosing build configuration, target and what-not.

glikely, 2025-04-19:
Build system has been entirely updated. Windows build should now match the obs-plugintemplate instructions.

closed 2025-04-19 by glikely