glikely/obs-ptz #161: renaming "device" to cam*

JanGVH, 2023-05-23:
in version 12.X I renamed the camera's from devicexxx to Cam1 and Cam2 etc. because the operator knows Cam1 and Cam2, these names are in the script. my operators have no technical knowledge, and "device" is puzzling them. I updated to the last version but could not rename the devices. how do I rename? should I go back to 12.X?

AllenVM, 2023-11-12:
We have the same problem, We really need the ability to rename the PTZ Device names to match what the OBS operator understands them to be.

glikely, 2025-04-30:
The device name is set to the source name the camera is associated with. Change the name of the source to change the displayed name of the camera.

closed 2025-04-30 by glikely