glikely/obs-ptz #162: Obs fecha ao salvar a porta usb

Oluizlopes, 2023-05-31:
Olá pessoal, o plugin está com bug quando seleciona a camera nas configurações e adiciona a porta usb o obs fecha, não consigo controlar a PTZ

glikely, 2023-06-01:
Hi @Oluizlopes 

USB cameras are not supported but the plugin, but it still should not cause OBS to crash. Can you please attache the OBS log file so that I can see what is going on?

rodto, 2023-06-06:
I had the same issue.

https://obsproject.com/logs/NA5ste8KWMQWJBCl

vishy-git, 2023-06-06:
If VISCA serial support is not discontinued, we should fix this. If not, drop support for this. Either way, an update to the plugin is needed. This seems to have broken after OBS 28.1.1 and 29.

normen, 2023-06-09:
Same issue here, OBS simply crashes when obs-ptz tries to connect to any serial port, no matter if USB, built-in or virtual.

sadale, 2023-06-10:
Same problem here.

bradleypeterson, 2023-06-20:
I'm seeing the same thing. In my case, it's a Fomako camera. Been using it for a couple of years on two machines without problems. I then updated a computer from Windows 10 to Windows 11. Now it crashes exactly when I select the COM port and then click "Apply"

According to the logs, this is what it looks like on a computer that accepts the camera after clicking "Apply":

`
22:41:02.882: [obs-ptz] plugin loaded successfully (version 0.16.0-test2)  
22:41:03.078: [obs-ptz] Looking for UART object COM3  
22:41:03.078: [obs-ptz] Creating new VISCA object COM3  
22:41:03.104: [obs-ptz] COM3 --> 88:30:01:ff 22:41:04.255: [obs-websocket] [obs_module_load] you can haz websockets (Version: 5.1.0 | RPC Version: 1)  
22:41:04.255: [obs-websocket] [obs_module_load] Qt version (compile-time): 6.3.1 | Qt version (run-time): 6.3.1  
22:41:04.255: [obs-websocket] [obs_module_load] Linked ASIO Version: 101201  
22:41:04.262: [obs-websocket] [obs_module_load] Module loaded.  
`

And this is a computer that gives the crash when clicking "Apply":

`22:44:30.090: [obs-ptz] Looking for UART object COM4   
22:44:30.090: [obs-ptz] Creating new VISCA object COM4`  

I've been able to rule out the camera as being an issue, as it works on another Windows 11 computer.

I've also been able to rule out USB->serial cable driver. On the computer that crashes in OBS, I installed this tool: https://ptz-controller.software.informer.com/3.9/, and it read camera no problem on COM4 and controlled it with VISCA commands.

Seems something is going wrong loading up the com port.

[log-working.txt](https://github.com/glikely/obs-ptz/files/11796282/log-working.txt)
[log-not-working.txt](https://github.com/glikely/obs-ptz/files/11796283/log-not-working.txt)

bradleypeterson, 2023-06-28:
I've narrowed it down to the PC causing the problems. this PC on Windows 11 using OBS-PTZ crashes when clicking "Apply" after selecting the COM port. All other configurations I've tried worked.

I was also able to narrow it down to a line of code. I got the plugin to compile on Windows. (Build notes need updating btw, I had to use the github build scripts.)   I updated the code below as follows

`
//In uart-wrapper.hpp
//... some code
bool PTZUARTWrapper::open()
{
	blog(LOG_INFO, "In PTZUARTWrapper::open(), preparing with uart.open(QIODevice::ReadWrite)",
		     qPrintable(port_name));
	bool rc = uart.open(QIODevice::ReadWrite);
	blog(LOG_INFO, "In PTZUARTWrapper::open(), finished with uart.open(QIODevice::ReadWrite)",
		     qPrintable(port_name));
//... more code		     
`
Sure enough, the log file shows "preparing with uart.open..." and it never reaches "finished with uart.open..."

That means it's crashing at 	
`bool rc = uart.open(QIODevice::ReadWrite);`

bradleypeterson, 2023-06-28:
As for the reason why I've narrowed it down to a PC, here's the details. I have two laptops, 2 cameras, and 4 different USB to serial cords. 

Computer A: HP pavilion
Computer B: Dell Inspiron

Both running OBS PTZ v0.16.0 codebase

Cord 1: Oikwan, claims to be FTDI chipset, but computer immediately gave PL2303TA chip error “PL2032TA DO NOT SUPPORT WINDOWS 11 OR LATER, PLEASE CONTACT YOUR SUPPLIER”
Cord 2: TRENDnet TU-S9 Prolific USB to serial. PL2303TA chip. Also generates PL2303TA chip error “PL2032TA DO NOT SUPPORT WINDOWS 11 OR LATER, PLEASE CONTACT YOUR SUPPLIER”
Cord 3: DTech uses FT232RL chipset
Cord 4: Keyspan Serial USB adapter, unknown chipset

-----------------------
Test setups and results
-----------------------

Computer A - Cord 1 - Fomako - Windows 11: **Worked**
Computer A - Cord 2 - Fomako - Windows 11: **Worked**
Computer A - Cord 3 - Fomako - Windows 11: **Worked**
Computer A - Cord 4 - Clear Touch RL400  - Windows 11: **Worked**

Computer B - Cord 1 - Fomako - Windows 10: **Worked**

_(I then updated Computer B to Windows 11)_

Computer B - Cord 1 - Fomako - Windows 11: **Did not work**
Computer B - Cord 2 - Fomako - Windows 11: **Did not work**
Computer B - Cord 3 - Fomako - Windows 11: **Did not work**
Computer B - Cord 4 - Clear Touch RL400 - Windows 11: **Did not work**

Computer B - Cord 1 - Fomako - Windows 11 - Used PTZ Controller 3.9: **Worked**
Computer B - Cord 2 - Fomako - Windows 11 - Used PTZ Controller 3.9: **Worked**

-----------------------
As you can see, the only scenario that repeatedly didn't work was Computer B + Windows 11 + OBS PTZ plugin.  Computer B can talk to the camera using another PTZ tool. Computer B could also talk to the camera in Windows 10 using OBS PTZ. It's just something about the three tuple of Computer B + Windows 11 + OBS PTZ plugin causing the problem.

At this point the only next step I can think of is to try and narrow down why QT6 crashes when trying to open any com port on computer B. I was able to get a QT6 hello world program to compile that uses the serial library, but merely including the library and doing nothing else seg faulted the program before it could even display "Hello world". So...I'm stuck and not sure how I can proceed.

vishy-git, 2023-06-29:
> 

To add to this, I have a computer with Windows 11, OBS 28.1.1 and PTZ Control 12.0 works. 

Upgrade OBS to 29.* and it breaks. 

Upgrade to PTZ Control 15.4 with OBS 28.1.1 and it breaks.

bradleypeterson, 2023-07-16:
> > 
> 
> To add to this, I have a computer with Windows 11, OBS 28.1.1 and PTZ Control 12.0 works.
> 
> Upgrade OBS to 29.* and it breaks.
> 
> Upgrade to PTZ Control 15.4 with OBS 28.1.1 and it breaks.

Yes, I'm seeing something similar.

OBS 28.1.1 + Windows 11 works on all PTZ Control versions, including a custom build from the latest source code.

OBS 29 still does not work. 

I did not install the earliest 29 betas to see when the problem cropped up. But at this point I wonder if it's an OBS bug and not a QT bug or a PTZ Control bug.

djsmacedo, 2023-08-21:
Same problem here with version 29. Working with version 28.

devinbost, 2023-09-09:
I experienced this as well on Windows 11 (not on my M1 Macbook Pro), and the event viewer entry named Qt6Core.dll in the crash. 
If it's crashing, that indicates an unhandled exception being thrown. 

@glikely 
If it's crashing at:
`bool rc = uart.open(QIODevice::ReadWrite);`
this is even more suspect.

From the docs, it appears that an error from an inability to open the device should show up here: https://doc.qt.io/qt-6/qserialport.html#error-prop

But, in the present case, that's not what's happening. Instead of returning false, it seems that the QSerialPort object is internally throwing an exception before it can return false (likely a bug in the library). 

We could try to upgrade the library, but that might cause other trouble if there are breaking changes, so I'd like to see the exception so I can dig a bit further and see if there's a known issue that's been fixed in a more recent version. 
What about trying something like this? 

```
    try {
        bool rc = uart.open(QIODevice::ReadWrite);
    } 
    catch(const std::exception& e) {
       blog(LOG_INFO, "Qt6 error when trying to open UART for VISCA (uart.open(QIODevice::ReadWrite))- Exception is %s",
		     qPrintable(e.what()));
        throw; 
    } 
```

devinbost, 2023-09-09:
I'm still working on getting my dev environment setup, but I started on this here. (Don't merge yet since I don't even know if it compiles yet.) https://github.com/glikely/obs-ptz/pull/174

bradleypeterson, 2023-09-11:
I recall putting in a try catch block, and it still crashed. (I unfortunately don't recall what datatype I was catching).

I also suspect it's a QT6 bug. I just couldn't get a QT6 hello world program to run without seg faulting. My hope was that if I could just have some small < 100 line program open up the serial port, then I'd have a much easier time narrowing it down further.

If you want any help on my end, I'm more than happy to test out anything you've got.

GustavoAntonny, 2023-09-25:
Hi guys, I was facing the same issue, you can fix it by adding this version (6.4.3.0) of QtSerialPort.dll to the OBS root folder, and the problem is gone! I hope it helps!

[Qt6SerialPort.zip](https://github.com/glikely/obs-ptz/files/12710528/Qt6SerialPort.zip)

devinbost, 2023-10-15:
Is this a version upgrade of QtSerialPort? I'm trying to find where the versions are specified, but I'm not sure how to translate the way the version is specified (https://github.com/glikely/obs-ptz/blob/main/buildspec.json) with the way they're defined in Qt6. (Our project says we're using 2022-08-02, but I can't find which Qt6 version that release date maps to.)

devinbost, 2023-10-16:
I found the old DLL in C:\Program Files\obs-studio\obs-plugins\64bit, and the metadata indicated it was version 6.3.1.0. That new file is version 6.4.3.0, and when I replaced the old version with the new one, it worked!!!

devinbost, 2023-12-03:
After the latest upgrade of OBS, I'm getting a version incompatibility. How do I get the new Qt6SerialPort.dll?
I'm not finding it after installing Qt6.

devinbost, 2023-12-03:
I found it by installing both the MinGW version of Qt and the Qt Serial Port module in Additional Libraries in the Qt6 installer. After that, it was in the mingw_64/bin directory.

However, I couldn't get the plugin to load with the new version.

techwizard8391, 2024-01-22:
Anyone achieves a resolution to this, as even after the file is placed in the root directory, after applying the camera settings OBS crashes.

glikely, 2024-02-25:
Hi everyone. Apologies, but I haven't been able to spend any time on this project in the last year. Yes, the problem is that OBS Studio updated the Qt6 libraries, but the qtserialport library shipped with the plugin is built against an older version. The way I've included the library is inherently fragile. Currently obs-ptz handles it by rebuilding all of obs-deps, and then packaging the resulting qtserialport. Fixing properly requires build system rework.

I might have to drop serial port support in the packaged versions

The ideal solution would be for OBS-Studio to build and ship qtserialport with the rest of qt6, but when I've asked for that to be done the patch has been rejected.

Regardless, sorry for not being able to do anything with this for the last year. I'll try to get it fixed, but I cannot make any promises.

Bluescooter42, 2024-03-08:
That you the update and honest.

Scott Lindsay

devinbost, 2024-03-15:
Is there a link to where they've rejected the ask to include qtserialport
in the official build? I don't understand why that would be rejected
upstream.

glikely, 2025-04-30:
The QSerialPort issue is now solved. It is building inside the source tree. Please try the latest prerelease and provide feedback on the discussion page:

https://github.com/glikely/obs-ptz/discussions/214

closed 2025-04-30 by glikely