glikely/obs-ptz #164: OBS (current) crashes after adding VISCA serial USB PTZ

EctoMonkey, 2023-06-14:
Try to add PTZ Tongveo (USB) to OBS remote via VISCA serial and COM Port. After clicking “Apply” OBS crashes.

bradleypeterson, 2023-06-20:
I'm seeing the same thing. In my case, it's a Fomako camera. Been using it for a couple of years on two machines without problems. I then updated a computer from Windows 10 to Windows 11. Now it crashes exactly at the "Apply" point you described.

According to the logs, this is what it looks like on a computer that accepts the camera after clicking "Apply":

`22:41:02.882: [obs-ptz] plugin loaded successfully (version 0.16.0-test2)
22:41:03.078: [obs-ptz] Looking for UART object COM3
22:41:03.078: [obs-ptz] Creating new VISCA object COM3
22:41:03.104: [obs-ptz] COM3 --> 88:30:01:ff
22:41:04.255: [obs-websocket] [obs_module_load] you can haz websockets (Version: 5.1.0 | RPC Version: 1)
22:41:04.255: [obs-websocket] [obs_module_load] Qt version (compile-time): 6.3.1 | Qt version (run-time): 6.3.1
22:41:04.255: [obs-websocket] [obs_module_load] Linked ASIO Version: 101201
22:41:04.262: [obs-websocket] [obs_module_load] Module loaded.`

And this is a computer that gives the crash when clicking "Apply":

`22:44:30.090: [obs-ptz] Looking for UART object COM4
22:44:30.090: [obs-ptz] Creating new VISCA object COM4`

I've been able to rule out the camera as being an issue, as it works on another Windows 11 computer.  

I've also been able to rule out USB->serial cable driver. On the computer that crashes in OBS, I installed this tool: https://ptz-controller.software.informer.com/3.9/, and it read camera no problem on COM4 and controlled it with VISCA commands.

Seems something is going wrong loading up the com port.

[log-working.txt](https://github.com/glikely/obs-ptz/files/11796258/log-working.txt)
[log-not-working.txt](https://github.com/glikely/obs-ptz/files/11796259/log-not-working.txt)

normen, 2023-06-20:
See #162 - you're not alone.

Sciloski, 2023-06-27:
Having the same issues as well, if try and change from visca tcp to any other protocol it crashes OBS.  I have tried adding and deleting the existing camera as well.

djsmacedo, 2023-08-21:
Same problem here with version 29. Working with version 28.

devinbost, 2023-09-09:
I ran into this issue on a Windows machine (not my Macbook), and when I checked the event viewer, the error specified Qt6Core.dll
(There wasn't a stack trace, unfortunately.)

GustavoAntonny, 2023-09-25:
I hope it helps! [QtSerialPort-6.4.3.0](https://github.com/glikely/obs-ptz/issues/162#issuecomment-1732805442)

hamsterbaitor, 2024-01-30:
I had this problem too. GustavoAntonny pointed me in the right direction, however the version of OBS I was using was built against qt 6.5.3 so I need to find the corresponding version of the serial port I'll (i.e. 6.5.3) and then it worked!

closed 2025-04-20 by glikely