glikely/obs-ptz #165: OBS PTZ controls crashed on OBS 29.1

NZSteven, 2023-07-04:
I was using OBS 29.0.2 with the OBS PTZ controls. Yesterday I upgraded the OBS to 29.1.3, the PTZ control crashed. Today I have trying all different OBS version and found the PTZ crashed after 29.1

Regards
Steven

rafael2437, 2023-08-19:
Também estou passando pelo mesmo problema. Espero que o problema seja corrigido nas próximas versões do OBS.
Obrigado.

djsmacedo, 2023-08-21:
Same problem here.

NZSteven, 2023-08-22:
> This plugin hasn't been updated since before OBS version 29 was released.
> I think there may have been changes to the targeted version of QT as part
> of version 29, which could be causing some of the issues. I am setting up a
> QT build environment to test if a more recent version of qtserialport will
> alleviate some of the issues.

Thanks

bpmacedo, 2023-09-07:
After updating OBS Studio to v29.1.3 it doesn't open. I realized that "obs-ptz" is causing the problem.

For me, it doesn't crash but I can see the last line on the log file is:
```
13:15:41.136: [obs-ptz] plugin loaded successfully (version 0.15.4)
```

After removing the plugin files I can open OBS again and see the line bellow on the log file
```
13:17:03.660: ==== Startup complete ===============================================
```

@glikely any plans to update the plugin?

----

> This plugin hasn't been updated since before OBS version 29 was released. I think there may have been changes to the targeted version of QT as part of version 29, which could be causing some of the [...]

@GeNTooGeek did you make any progress? Can I help anyway?

GeNTooGeek, 2023-09-08:
Yes and no. This has been my first foray into QT and after a few trials to get the right version for the obs-studio source I had, I was able to build a worming copy of the plugin. But, I then realized the current master branch has moved to 30.0, not 29.1 and the plugin I built only worked on 30.0.

I ran out of time that night to redo my efforts and it looks like it will be into October before I will have time to try again.

So, while I don't have a working copy of the plugin that will work, I believe a new build of the QtSerialPort dll and the plugin with the appropriate 29.1 branch code will work.

GustavoAntonny, 2023-09-25:
I hope it helps! [QtSerialPort-6.4.3.0](https://github.com/glikely/obs-ptz/issues/162#issuecomment-1732805442)

GeNTooGeek, 2023-10-04:
Yes, thank you.  It was nice to not have to try re-compiling that.  That version of the QTSerialPort works with 29.1.3.  Just a note from the other issue.  You can place the dll in the plugins/64bit folder along with the other plugin folders instead of in the obs-studio root fold.er

hamsterbaitor, 2024-01-30:
I had this problem too. GustavoAntonny pointed me in the right direction, however the version of OBS I was using was built against qt 6.5.3 so I need to find the corresponding version of the serial port I'll (i.e. 6.5.3) and then it worked!

bpmacedo, 2024-04-01:
So, it's been a while...

@GustavoAntonny thx 4 the link. I followed and it worked! OBS PTZ works with OBS 29.1.3

But new releases are out and we have the problem pointed by @hamsterbaitor 

- OBS 30.0.2 uses QT 6.5.3.0
- OBS 30.1.1 uses QT 6.6.2.0

@GeNTooGeek do you know where to get new versions of Qt6SerialPort.dll?

Thx in advance

hamsterbaitor, 2024-04-01:
@bpmacedo I had some google searching, and found some compiled versions on the msys2 website (specifically here: https://packages.msys2.org/package/mingw-w64-x86_64-qt6-serialport). It wasn't immediately obvious to me which way to capture a specific build, so I manually edited the download link to a relevant one (in this instance https://mirror.msys2.org/mingw/mingw64/mingw-w64-x86_64-qt6-serialport-6.6.2-1-any.pkg.tar.zst) and it seemed to provide me with the compiled dll at the right version. However, I've since had issues with 30.1.1 generating some NVIDIA conflicts that I need to resolve before I can test if it works!

Tsaukpaetra, 2024-05-22:
As a workaround, I personally was able to use a component of the com0com project to "tunnel" a local TCP port to the Serial port, specifically the [com2tcp program](https://sourceforge.net/projects/com0com/files/com2tcp/1.3.0.0/com2tcp-1.3.0.0-386.zip)

It looks like this:
[image]

From there you can just connect to localhost like so, and it should just work transparently:
[image]

Not sure about redistribution licensing, but it might be possible to process out to this program to make it slightly more seamless but this way you don't have a dependency on Qt that has to be recompiled for every version  of Obs.

bpmacedo, 2024-06-17:
@hamsterbaitor, thx 4 the links!

I just tested Qt6SerialPort.dll v6.5.3.0 with OBS 30.0.2 and v6.6.2.0 with OBS 30.1.2 but it didn't worked 😢 

the log file shows the error below

```
Module '../../obs-plugins/64bit/obs-ptz.dll' not loaded
```

Maybe, It's necessary to update the other files of the plugin (Qt6SerialPortd.dll, Qt6SerialPortd.pdb) to work.

---
I didn't test the workaround Tsaukpaetra pointed out because It seems to be a little complicated to use during the events

I think OBS 29.1.3 is the end of the line for the obs-ptz plugin until someone else finds a way to update the DLL files

hamsterbaitor, 2024-06-17:
Thanks for the feedback @bpmacedo . 

I think the workaround from @Tsaukpaetra will also be an issue as OBS isn't recognising the plugin at all, so can't fall back on VISCA over IP.

I'll try and have a look about building the plugin from source using updated Qt and see if that works, but I'm very much pushing the limits of my technical knowledge on this!

bpmacedo, 2024-06-17:
@hamsterbaitor please let me know if I can help with this...

have you checked this other repo https://github.com/glikely/obs-deps-qtserialport ?

bpmacedo, 2024-06-20:
> Some news!
> 
> I used the files Qt6SerialPort.dll, Qt6SerialPortd.dll, Qt6SerialPortd.pdb from the repo above (https://github.com/glikely/obs-deps-qtserialport/releases/download/2023-11-03-serialport/windows-deps-qt6-2024-01-03-x64.zip), and the plugin is now loaded in the OBS v30.0.2. I can't test it right now because the camera is turned off and I'm accessing the PC remotely

hi!
I just wanted to let you know that I tested the camera with these files above and it's working on OBS v30.0.2 👍 

Unfortunately, it doesn't work with OBS v30.1+
The message below appears in the log file
```
Module '../../obs-plugins/64bit/obs-ptz.dll' not loaded

GeNTooGeek, 2024-10-06:
Here is a compiled version of QT6SerialPort that should work with obs-ptz and OBS 30.2.3

[Qt6SerialPort-QT6.6.3-OBS30.2.3.zip](https://github.com/user-attachments/files/17270706/Qt6SerialPort-QT6.6.3-OBS30.2.3.zip)

NZSteven, 2024-10-06:
Thanks. Where should I copy the dll file to? the folder of the PTZ control,
or Windows?

Steven

GeNTooGeek, 2024-10-06:
Same as the plugin.

GeNTooGeek, 2024-10-17:
The .pdb file is the debug database for the dll. The other one was an artifact of certain build configuration options, and wasn't needed when I found the right ones.

@GeNTooGeek<https://github.com/GeNTooGeek> thx for sharing with us!
I tested and can confirm that it's working with obs 30.2.3

What do u think about files Qt6SerialPortd.dll and Qt6SerialPortd.pdb? IDK if it's important to update them, too.

imiddleton1, 2024-11-04:
I just updated OBS to 30.2.3 and tried the DLL shared by [GeNTooGeek](https://github.com/GeNTooGeek) but I am still getting crashes in obs-ptz.dll when changing scene collections.
[Crash 2024-11-04 16-36-15.txt](https://github.com/user-attachments/files/17624109/Crash.2024-11-04.16-36-15.txt)

bpmacedo, 2024-11-05:
@imiddleton1 I don't use other scene collection but I duplicated the one that I use to test what you said and I can't reproduce the problem... IDK if I'm doing it like you but here I could change between scene collections without problem.

Before I got the DLL from GeNTooGeek I updated other files and posted on a comment... I hope it can help you

> Some news!
>
> I used the files Qt6SerialPort.dll, Qt6SerialPortd.dll, Qt6SerialPortd.pdb from the repo above [...]

dpcavalcanti, 2024-11-21:
> As a workaround, I personally was able to use a component of the com0com project to "tunnel" a local TCP port to the Serial port, specifically the [com2tcp [...]

Hey @Tsaukpaetra I am trying this workaround you used but no success, could you explain how you did that work? im my case, the plugin start conecct and disconnect repeatedly.

bpmacedo, 2025-04-24:
> [@GeNTooGeek](https://github.com/GeNTooGeek) thx for sharing with us! I tested and can confirm that it's working with obs 30.2.3

I want to add more info here:
On 04/18, I updated OBS to version 31.0.3 with this compiled version, and the plugin works great!

glikely, 2025-04-24:
I've got the plugin working now against ONS 31.0.3. I'm not ready to make a release as there are still a lot of problems, but if anyone wants to give it a go you can find the binaries here:

https://github.com/glikely/obs-ptz/actions/runs/14649945900

Unfortunately I don't yet have a qserialport solution, so serial port VISCA is disabled in that build, as is joystick support.

closed 2025-04-30 by glikely