glikely/obs-ptz #168: obs-ptz causes OBS 29.1.3 to sometimes crash when switching scenes

Elmoria, 2023-07-26:
When I switch scenes in OBS, the programme frequently crashes. The OBS crash log shows that obs-ptz is probably causing this issue.
Here is one of the crash logs:
[Crash 2023-07-26 09-28-31.txt](https://github.com/glikely/obs-ptz/files/12169274/Crash.2023-07-26.09-28-31.txt)

The problem does not only occur with one specific scene, nor does it always occur. I can't find a pattern.

I tried reinstalling everything on different hardware but I still have the same problem.
I also tried putting obs-ptz in debug mode. But nothing helpful was logged for this particular problem.

If I can help in any way to solve this problem, or if more information is needed, please let me know. 

Thanks in advance!

AlexH269, 2023-08-09:
I am facing the same problem with our local church.

glikely, 2025-04-20:
This is another symptom of the latest release (v0.15.4) only working with OBS Studio 28. I'm closing this as a duplicate as it is tracked in another issue.

closed 2025-04-20 by glikely