glikely/obs-ptz #169: Custom visca commands

phill-85, 2023-08-06:
I would like to see some buttons next to the focus ones which can have custom visca commands.

My use case would be a button to turn the Ai tracking on and off on my ptz camera. Using the following commands.

markusd1984, 2023-11-03:
custom commands could help to cover any other scenarios supported by cameras indeed.

I've added a button for AI tracking on/off , will try to share at some point, need to learn more how to propose a commit, otherwise I can share my code but it needs local compiling i think to work as I couldn't yet manage to create a deployable one.

https://github.com/glikely/obs-ptz/issues/155

jerano, 2026-02-23:
I'd like to second this, as a "Feature Request".
I know that ie. Cisco PrecisionHD cameras need to be sent an "IR Disable" command, to be able to modify the Pan/Tilt speeds.

glikely, 2026-10-02:
VISCA commands are now stored in json visca profile files. This allows new controls to be added, but there isn't any UI for it yet. Still, progress and this issues has not been forgotten. Stay tuned