glikely/obs-ptz #172: Activating hotkey by ID crashes obs-ptz

jbrown123, 2023-09-03:
I'm using [Bitfocus Companion](https://bitfocus.io/companion) to remote control OBS. It can generate keystrokes or it can activate a hotkey registered with OBS via the "activate hotkey by ID" API. I believe they are using the websocket API to communicate with OBS and calling the TriggerHotkeyByName function.

When I attempt to activate a PTZ preset (e.g. PTZ.Recall1) via the bitfocus tool I'm able to do that once or twice, then no presets work. I can't activate a preset using the mouse on the interface in OBS, an assigned hotkey, etc. Basically the PTZ plugin is just "dead" at that point. I have to restart OBS to get it to work again.

If I instead tell Companion to trigger the hotkey by keypress (e.g. control-alt-shift-1) then it works just fine.

I do this same thing (use Companion with the hotkey by ID) with OBS commands and with other plugins and they work fine. Notably the lower third plugin for example (https://obsproject.com/forum/resources/animated-lower-thirds-with-dockable-control-panel.1057/). This leads me to believe it is something in the obs-ptz plugin that is having issues.

normen, 2023-11-22:
It might have to do with the fact that the plugin needs the "key released" to know when to stop controlling camera movement (given, thats not the case for presets). The websocket commands are just "triggers", so theres no key release event.

glikely, 2025-04-30:
Please try again with latest prerelease and reopen if you still have the issue

closed 2025-04-30 by glikely