glikely/obs-ptz #173: Power off camera through visca

GarthDB, 2023-09-04:
All of the Visca over IP command lists I can find show an option for powering off the camera (or standby for BirdDog). I noticed the plug-in has an option for powering on the camera, but it would be nice to power it off as well. 

[image]

I imagine it as a checkbox option in the config to send the power off command to the camera when OBS shuts down. If I'm reading the OBS plug-in api docs correctly there is a [function](https://docs.obsproject.com/reference-core#c.obs_shutdown) that could be used to handle this.

I'm happy to put together a pr if it's likely to get merged. The use case for me is we have a permanently installed camera that is used once a week and is just left on the rest of the time. It would be nice to save a little power and maybe prolong the life of the camera by being in standby or turned off.

markusd1984, 2023-11-03:
This is so needed for churches and similar situations, if only the camera supports the command though :) 

Another way could be custom VISCA Command buttons https://github.com/glikely/obs-ptz/issues/169

I had to modify the plugin to add a button for turning on/off Auto tracking.

ogmkp, 2024-01-08:
But I have the Power off right click option on my side (after clicking on Power ON who don't know the real state of the camera).

glikely, 2025-04-30:
Power on and off are both supported.

closed 2025-04-30 by glikely