glikely/obs-ptz #175: Obs 29.1.3 with PTZ Controls crashes

corys70155, 2023-09-18:
Obs 29.1.3 with PTZ Controls crashes every time I try to put in a camera control. I am using a Waveshare rs-485 USB adaptor to connect with the cameras. These cameras do not have an IP address. They use a com port and a Device ID but when you hit "Apply" OBS just shuts down. Is there a solution for this or should I look for another way to connect?
[2023-09-18 12-41-55.txt](https://github.com/glikely/obs-ptz/files/12653278/2023-09-18.12-41-55.txt)
[2023-09-18 12-04-47.txt](https://github.com/glikely/obs-ptz/files/12653280/2023-09-18.12-04-47.txt)
[2023-09-18 10-45-07.txt](https://github.com/glikely/obs-ptz/files/12653282/2023-09-18.10-45-07.txt)
[2023-09-15 09-50-28.txt](https://github.com/glikely/obs-ptz/files/12653285/2023-09-15.09-50-28.txt)
[2023-09-15 09-52-37.txt](https://github.com/glikely/obs-ptz/files/12653286/2023-09-15.09-52-37.txt)
[2023-09-17 08-45-48.txt](https://github.com/glikely/obs-ptz/files/12653287/2023-09-17.08-45-48.txt)
[2023-09-17 13-24-20.txt](https://github.com/glikely/obs-ptz/files/12653288/2023-09-17.13-24-20.txt)
[2023-09-18 10-31-19.txt](https://github.com/glikely/obs-ptz/files/12653289/2023-09-18.10-31-19.txt)
[2023-09-15 09-34-05.txt](https://github.com/glikely/obs-ptz/files/12653290/2023-09-15.09-34-05.txt)
[2023-09-15 09-43-26.txt](https://github.com/glikely/obs-ptz/files/12653291/2023-09-15.09-43-26.txt)

GustavoAntonny, 2023-09-25:
I hope it helps! [QtSerialPort-6.4.3.0](https://github.com/glikely/obs-ptz/issues/162#issuecomment-1732805442)

hamsterbaitor, 2024-01-30:
I had this problem too. GustavoAntonny pointed me in the right direction, however the version of OBS I was using was built against qt 6.5.3 so I need to find the corresponding version of the serial port I'll (i.e. 6.5.3) and then it worked!

closed 2025-04-20 by glikely