glikely/obs-ptz #177: Add option to disable sequence for VISCA UDP

neonconsultingllc, 2023-10-13:
Some PTZ cameras seem to not support UDP packets that contain the sequence part of visca.
One example is the AKVANS AV-E20-NDI. 
I also have some LinkPi devices which can forward udp packets to a serial (visca) port, which does not work correctly when sequence part of packet is used.

For testing, I created a quick python script which removes anything before x81, then forwards on to the device.
```
import socket
import codecs
UDP_IP = "192.168.1.49"
UDP_PORT = 5000
sock = socket.socket(socket.AF_INET, socket.SOCK_DGRAM)
sock.bind((UDP_IP,UDP_PORT))
CAM_IP = "192.168.1.45"
CAM_PORT = 1259
cam_sock = socket.socket(socket.AF_INET, socket.SOCK_DGRAM)
while True:
    data, addr = sock.recvfrom(2048)

    d=codecs.getencoder('hex')(data)
    s = None
    x = list()
    for i in range(0,d[1]):
        if hex(data[i]) == '0x81':
            s = i
        if s is not None:
           x.append(data[i])
    send = bytearray(x)
    print("%s -> %s" % ( data,send))
    cam_sock.sendto(send,(CAM_IP,CAM_PORT))

```

Quick example of output:
```
b'\x01\x00\x00\t\x00\x00\x00E\x81\x01\x06\x01\n\x08\x02\x02\xff' -> bytearray(b'\x81\x01\x06\x01\n\x08\x02\x02\xff')
b'\x01\x00\x00\t\x00\x00\x00F\x81\x01\x06\x01\x0b\t\x02\x02\xff' -> bytearray(b'\x81\x01\x06\x01\x0b\t\x02\x02\xff')
b'\x01\x00\x00\t\x00\x00\x00G\x81\x01\x06\x01\x00\x00\x03\x03\xff' -> bytearray(b'\x81\x01\x06\x01\x00\x00\x03\x03\xff')
b'\x01\x00\x00\t\x00\x00\x00H\x81\x01\x06\x01\x01\x01\x01\x01\xff' -> bytearray(b'\x81\x01\x06\x01\x01\x01\x01\x01\xff')
b'\x01\x00\x00\t\x00\x00\x00I\x81\x01\x06\x01\x02\x02\x01\x01\xff' -> bytearray(b'\x81\x01\x06\x01\x02\x02\x01\x01\xff')
b'\x01\x00\x00\t\x00\x00\x00J\x81\x01\x06\x01\x04\x03\x01\x01\xff' -> bytearray(b'\x81\x01\x06\x01\x04\x03\x01\x01\xff')
b'\x01\x00\x00\t\x00\x00\x00K\x81\x01\x06\x01\x05\x04\x01\x01\xff' -> bytearray(b'\x81\x01\x06\x01\x05\x04\x01\x01\xff')
b'\x01\x00\x00\t\x00\x00\x00L\x81\x01\x06\x01\x06\x05\x01\x01\xff' -> bytearray(b'\x81\x01\x06\x01\x06\x05\x01\x01\xff')
b'\x01\x00\x00\t\x00\x00\x00M\x81\x01\x06\x01\x07\x06\x01\x01\xff' -> bytearray(b'\x81\x01\x06\x01\x07\x06\x01\x01\xff')
b'\x01\x00\x00\t\x00\x00\x00N\x81\x01\x06\x01\x08\x07\x01\x01\xff' -> bytearray(b'\x81\x01\x06\x01\x08\x07\x01\x01\xff')
b'\x01\x00\x00\t\x00\x00\x00O\x81\x01\x06\x01\n\x08\x01\x01\xff' -> bytearray(b'\x81\x01\x06\x01\n\x08\x01\x01\xff')
b'\x01\x00\x00\t\x00\x00\x00P\x81\x01\x06\x01\x0b\t\x01\x01\xff' -> bytearray(b'\x81\x01\x06\x01\x0b\t\x01\x01\xff')
b'\x01\x00\x00\t\x00\x00\x00Q\x81\x01\x06\x01\x0c\n\x01\x01\xff' -> bytearray(b'\x81\x01\x06\x01\x0c\n\x01\x01\xff')
b'\x01\x00\x00\t\x00\x00\x00R\x81\x01\x06\x01\x00\x00\x03\x03\xff' -> bytearray(b'\x81\x01\x06\x01\x00\x00\x03\x03\xff')
b'\x01\x00\x00\x05\x00\x00\x00S\x81\x01\x06\x04\xff' -> bytearray(b'\x81\x01\x06\x04\xff')
```

It would be great to have an option to "Disable sequence" when configuring a camera to use VISCA UDP, which drops anything before x81

normen, 2023-11-22:
Can confirm, given the bugs in the serial port implementation and some of my cameras having some specialties in the VISCA implementation anyway I created a bridge app so I can massage the data. The easiest way in your case I suppose would ba making a super small app that just opens a TCP server and connects that to the camera UDP, which is basically what I am doing. The TCP-Visca that OBS-PTZ outputs is basically what you want.

djlane, 2025-04-07:
Is this problem fixed? Trying to use a Avkan av-e20-ndi and it's not working but I can use packsender to send visca code no problem.

PR #217 (2025-04-30): Quirk: Support VISCA UDP without sequence field
Some cameras don't use the UDP sequence number field in their VISCA UDP protocol. Support them by adding a quirk flag to modify the behaviour.

Fixes #177
Closes #178
commit a6550cf0 (2025-04-30): Quirk: Support VISCA UDP without sequence field

Some cameras don't use the UDP sequence number field in their VISCA UDP
protocol. Support them by adding a quirk flag to modify the behaviour.

Fixes #177
Closes #178
closed 2025-04-30 by glikely