glikely/obs-ptz #180: Extending PTZ plugin to support AI functionality and automation

devinbost, 2023-10-29:
I'd like to have some way to subscribe to events from this plugin like when a camera preset is saved or when a preset is renamed so I can take actions, either programmatically or with the advanced scene Switcher. 

Case 1 - subscribe to PTZ actions:
If someone renames a preset from one of the default names, I want to generate a new scene that contains an action to move the camera to that preset.

Case 2 - programmatically trigger PTZ actions:
I'd like to be able to extend this plugin to where I can use custom python code or Advanced Scene Switcher actions to move cameras. 
For example, I can use AI to detect when the preview camera has a person whose head is off the screen. In this case, I want to trigger the PTZ action to move that camera until the AI detects the person as being back on screen. I can easily do the AI part, but I need a way to trigger the PTZ actions.

How would I do either of these with this plugin? I'm happy to contribute code but just need to know where to start.

normen, 2023-11-20:
I do this by simply running the detection in a separate app and sending VISCA to the camera directly. You also need to have a way to get the camera image which is out of the scope of this plugin anyway. I do that via NDI and webcam emulation.

glikely, 2025-04-30:
PTZ Controls do export an API. The Face tracker plugin uses it.

https://obsproject.com/forum/resources/face-tracker.1294/

closed 2025-04-30 by glikely