glikely/obs-ptz #183: Issue loading presets & zoom controls sticking

kren24, 2024-01-03:
When loading presets, it's necessary to click the saved preset 5 times before it loads. The manual controls stick when zooming and fails to stop (continues zooming after releasing click) - I've tried on two different installs (Win 10 & Win 11) with the same issues. Is this an issue with the camera or a configuration issue?

CimCup, 2024-02-08:
I have the same problem. 

Tested with
Tenveo PTZ Camera
PTZ Control Plugin 0.15.4 and 0.16.test2

RobActive, 2024-03-13:
First, Thank you for all the efforts you've put into developing this plugin. I have the same problem Tenveo HD Conference Camera 20x Version V5.32 firmware new-VHD-RK-20230511.bin . OBS Studio 30.0.2 (64 bit). Connected the camera to the PTZ controls plugin using VISCA/TCP port 52381. I can control the camera using the up/down quite reliably. However, when I zoom in or out, it will start and will not stop as expect after releasing the software’s button. Also, I cannot reliably set Presets. “If” a preset does work, it will not activate unless you quickly press the preset’s software button about 8 time instead of at single click.