glikely/obs-ptz #188: Fatal Error Causing OBS Crash

dexterdesign23, 2024-03-03:
When clicking "Apply" when adding a camera, OBS experiences a fatal error and completely closes out. When relaunching, it prompts to go into safe mode.

Sys info:
Windows 11 Build 22631
AMD Ryzen 5 5600G
32 GB RAM
NVIDIA GeForce GTX 1660 SUPER

imiddleton1, 2024-03-25:
I'm having the same issue.  

Here is a crash log from my system:
[Crash 2024-03-25 11-55-50.txt](https://github.com/glikely/obs-ptz/files/14747040/Crash.2024-03-25.11-55-50.txt)

junao21, 2024-04-10:
estou com o mesmo problema, nas novas versões da o erro e abre o modo de segurança. 
estou usando uma versão mais antiga que não ocorre esse problema (versão 29 do obs)

Tsaukpaetra, 2024-05-22:
Error is because apparently it is compiled with Qt 6.3.1 and OBS is using a different version (in my case 6.5.3). Not sure how to fix that on our side unless someone wants to try recompiling it?

snisonger, 2024-07-11:
I am having the same issue. Any updates?

hdfhh1, 2024-12-29:
Any updates to this issue yet? it's the last issue i would like to get working for my church streaming.

GralfR, 2025-01-12:
With OBS 28.1.2 plugin 0.15.4 is working, with OBS 31.0.0 crash on Windows 10 here, too.

closed 2025-04-22 by glikely