glikely/obs-ptz #192: Presets take 6 clicks to invoke.

Rakenvald, 2024-06-03:
I have saved the presets through the cameras gui and through the OBS plugin however you need to click on the preset 7-8 times to get it to act accordingly.

lightsaberluke, 2024-07-02:
I have the same issue but only have to click it 4 times.

Rakenvald, 2024-07-02:
> I have the same issue but only have to click it 4 times.

It seems you have fixed half of my issue.

lightsaberluke, 2024-07-02:
I have gone through multiple versions and it still has the same issue. Also, when trying to use "PTZ action" as a source to control it it does not change the preset. What camera are you using? Mine is a Tenveo.

Rakenvald, 2024-07-02:
A "Tongveo" Brand.

BenRommel, 2024-07-22:
I also have a Tongveo brand PTZ camera and have had issues not only with with multiple click to activate several different functions (HOME being just one) but also intermittent control issues. The camera seems to be a common thread.

glikely, 2025-04-21:
I suspect this is indeed a camera issue. Turn on debug logging in the plugin and reproduce the problem. Then attach the OBS log file to this issue please.