glikely/obs-ptz #197: Add ONVIF control of  Plugin  for network PTZ Controllers

Nwachinyere, 2024-07-16:
Allow hardware control of plugin through ONVIF, with the option to set port  number, username and password, and a feed of current PTZ Camera being controlled with name, or Just the name and IP address.

zapy85, 2024-09-02:
This would be a great feature for those cameras that does not support VESCA like Hikvision for ex.
I have a camera available if you would like to give this a try, the onvif protocol is available so i don't think it would be that hard to implement,

glikely, 2025-04-22:
Clarification question; are you asking for ONVIF camera support? Or control of the plugin with an ONVIF joystick?

zapy85, 2025-04-22:
> Clarification question; are you asking for ONVIF camera support? Or control of the plugin with an ONVIF joystick?

I would say proper onvif support, that would create endless opportunities in support of cameras.
As it is standardized it shouldn't be hard to implement, basic control as well for pan tillt zoom is like onvif profile s the lowest one.
Some cameras also have a microphone input that can be utilized as well to capture sound.
All sent over the rtsp stream.

https://www.onvif.org/wp-content/uploads/2019/12/ONVIF_Profile_-S_Specification_v1-3.pdf

Hardware control with a 3D connexion for ex would've been nice as well I guess if you're supporting like PTZ.

Nwachinyere, 2025-04-22:
The original request was for onvif controllers to be able to see and control the plugin like it does cameras. In this scenario the plugin is acting as a middle man allowing hardware control of the PTZ camera active on the scene.

glikely, 2025-04-30:
I'm not even sure how to go about this, not having an ONVIF controller myself. This would need someone to volunteer to work on it. I'm going to close this as not planned. Reopen if you are able to work on it, or can recruit someone who will.

closed 2025-04-30 by glikely

zapy85, 2025-04-30:
Regarding onvif support in the plugin should I open a new ticket or should
i direct that against OBS directly maybe?

glikely, 2025-04-30:
The plugin already has ONVIF camera support. It's not very good yet, but it is there.