glikely/obs-ptz #198: Install plug-in on Ubuntu Studio 24, OBS Studio 30

BenRommel, 2024-07-22:
OBS Studio 30.0.2.1-3build (64bit) bundled with Ubuntu Studio 24. Unable to get the plug-in to load. Running the .deb with Package Install results in Error: Cannot satisfy dependencies so installed in terminal window with:
**sudo dpkg -i obs-ptz-0.16.0-test2-linux-x86_64.deb**
Looks to go through the install process but then... nothing. More suggestions? Or just not compatible with Ubuntu Studio distro? Thanks.

jordanp123, 2024-09-14:
I'm with you, I cant' get it to work no matter what I do.

mandle420, 2024-09-16:
A copy of the terminal output here would help. Can't say much about "cannot satisfy dependancies". That could any number of things.  I'd also be nice to know which dependancies. 
You could also try installing the stable version, not testing. 
https://github.com/glikely/obs-ptz/releases/tag/v0.15.4

jordanp123, 2024-09-16:
Maybe I need to create my own thread (not sure) but I don't get the dependency message but when opening up OBS and checking the docks section there is nothing, no option for the PTZ dock. I've tried multiple times now with the same results. Maybe I'm just missing something although I don't know what.

jordanp123, 2024-10-01:
Looked at the log files (after taking a break) and installing libqt6serialport6 from the apt and now everything is working great.

closed 2025-04-20 by glikely