glikely/obs-ptz #202: obs-ptz-controls wont install from aur

mandle420, 2024-09-16:
I believe the issue is here. if it's just an arch issue, I've made a report on the aur page as well. 
using the temp fix from the manjaro forums, the first cmake section here shows up as well. SDL2 is installed and working.

....
CMake Warning at CMakeLists.txt:107 (find_package):
  Found package configuration file:

    /usr/lib/cmake/SDL2/SDL2Config.cmake

  but it set SDL2_FOUND to FALSE so package "SDL2" is considered to be NOT
  FOUND.
....
cc1plus: all warnings being treated as errors
make[2]: *** [CMakeFiles/obs-ptz.dir/build.make:129: CMakeFiles/obs-ptz.dir/src/ptz-controls.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
cc1plus: all warnings being treated as errors
make[2]: *** [CMakeFiles/obs-ptz.dir/build.make:311: CMakeFiles/obs-ptz.dir/src/imported/properties-view.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:86: CMakeFiles/obs-ptz.dir/all] Error 2
make: *** [Makefile:156: all] Error 2
==> ERROR: A failure occurred in build().
    Aborting...
 -> error making: obs-ptz-controls-exit status 4
 -> Failed to install the following packages. Manual intervention is required:
....

Full term output here.
https://pastebin.com/RFWN8ZfM

Temp fix from manjaro forums
https://forum.manjaro.org/t/obs-ptz-controls-fails-to-build/162227/13

git clone https://aur.archlinux.org/obs-ptz-controls.git
cd obs-ptz-controls
makepkg -o
sed -i '177 s/./#&/' src/obs-ptz/cmake/ObsPluginHelpers.cmake
makepkg -e

Not quite sure how to make it work, but at least it installs this way.(still need to rtfm) Once I have a chance to configure and test and i'll make an update. The person on the manjaro forums said they made a report, so if this is a repeat, my apologies. But I could not find it.

tytan652, 2024-09-20:
I took the time to dive into that warning, seems like obs-ptz buildsystem ask for a static build of SDL2 where Linux distributions are more likely to provide only a shared build.

ryzendew, 2024-10-03:
it worked fine

glikely, 2025-04-21:
Fixed in #210

closed 2025-04-21 by glikely