glikely/obs-ptz #203: Cannot install obs-ptz-0.15.4-linux-x86_64.deb with OBS-Studio flatpak

dhgarcia, 2024-10-09:
When trying to install obs-ptz:
`sudo dpkg -i obs-ptz-0.15.4-linux-x86_64.deb`

I get an error Package obs-studio is not installed.

I have OBS studio installed with flatpak  

Is there a way to make it recognise the flatpak installation?

herbertbf, 2024-11-28:
> When trying to install obs-ptz: `sudo dpkg -i obs-ptz-0.15.4-linux-x86_64.deb`
> 
> I get an error Package obs-studio is not installed.
> 
> I have OBS studio installed with flatpak
> 
> Is there a way to make it recognise the flatpak installation?

Did you manage to solve your problem? I can install it via the .deb package, but the plugin does not appear in the OBS

glikely, 2025-04-20:
The release is woefully out of date and the ubuntu packages never really worked properly anyway. I'm closing this issue as most of the problems like this need to be retested now that the build infrastructure has been updated to the latest obs-plugintemplate.

closed 2025-04-20 by glikely