glikely/obs-ptz #229: Bug: follow preview camera doesn't work

jg00dman, 2025-05-19:
There is a little button for selected camera to follow preview. This seems not to function.

glikely, 2025-05-19:
In the camera settings, do you have the camera connected to a source? That is needed to be able to switch between active cameras when the scene changes.

glikely, 2025-05-19:
However, the autoselect feature has been reworked and that button will be gone in the next release. I’m closing this issue because it is no longer relevent

closed 2025-05-19 by glikely

jg00dman, 2025-05-20:
Ah thanks, that did resolve it for me. I hope we aren't losing that feature? It's very handy!

glikely, 2025-05-20:
The feature is still there, it’s just been reworked. Now it shows up in the context menu when you right click on the list.

v0.17.0 was released yesterday with the new interface. Give it a try.