glikely/obs-ptz #235: Logitech Extreme 3D Pro Mapping

Spikey84, 2025-05-22:
When attempting to use a Logitech Extreme 3D Pro as a joystick controller, we seem to be unable to use many of the buttons and axes. To what attent would there be support available here for specific controllers? Should I make an issue with the joystick library? Thanks!

glikely, 2025-05-22:
What version of the plugin are you using?

Spikey84, 2025-05-22:
I am using it at work and I was not the one who downloaded it. I just asked and will get back to you asap. Thanks for the quick support!

Spikey84, 2025-05-22:
Just confirmed we were running 0.17!

Spikey84, 2025-05-22:
To add some specific information in hopes of potentially making my goals more clear, I am going to use a diagram for a flight sim game as an example to make it clear which button/axis I am concerned with. Here is a link to that chart:  [Flight Sim Map](https://www.reddit.com/r/MicrosoftFlightSim/comments/id2729/logitech_extreme_3d_pro_controller_map_cheat_sheet/). We are currently able to use the main left-right movement for pan and the main forward-back for tilt. We would like to be able to invert our tilt. Is that possible? Would you like that in a separate issue? Our last important need is for a zoom control. We were able to map the Axis labeled as throttle (on the example map), but it is largely impractical since it maintains its level rather than moving back to 0. We would like to use the hat up-down for zoom, but the hat does not seem to show up as an axis at all. It also does not show up as a set of buttons, which I was quite confused by. Could another piece of software using the controller disrupt the data from going into ptz control? Even once the zoom issue is resolved, we would really like to be able to use the rest of the buttons for preset looks. The season for our venue starts 6/7, so we are not in much of a rush, but if there is anything else I can do to help resolve this, please let me know. I am always happy to provide more details!

glikely, 2025-05-22:
I’m already planning to implement axis inversion, so you should have that soon.

The hat is a bit harder. It is reported as a 360 degree angle instead of a button press by the joystick library I’m using. I’m unlikely to get it wired up anytime soon.

Mapping buttons to presets is coming, but no promises on when.

Spikey84, 2025-05-22:
As an alternative to the hat, could zoom easily be mapped to buttons? Like one does zoom out and one does zoom in? Thank you so much for the help!

Spikey84, 2025-05-23:
Also, could the rotation of the stick be mapped as an axis?

commit f0406a29 (2025-05-31): joystick: Add button control of pan/tilt/zoom/focus

Adds the ability to map axis actions (pan, tilt, zoom and focus) to
buttons. This commit also enables translation of the action strings.

Fixes: #235
closed 2025-05-31 by glikely