glikely/obs-ptz #262: Module '../../obs-plugins/64bit/obs-ptz.dll' not loaded

allanhanan, 2025-07-06:
I installed the latest version v18.0.1 of the plugin and added it to obs (v31.0.4) in the ProgramData plugins location. It didnt show up in the docks menu. I checked the log and it said
```
09:53:35.539: Module '../../obs-plugins/64bit/obs-ptz.dll' not loaded
```

I checked the obs-ptz.dll with dependency walker and the Qt6Core, Qt6Widgets and Qt6Network dll files are loaded but are missing imports. I tried the dlls that came with obs itself, new dlls specified in buildspec.json
```
"qt6": {
            "version": "2024-09-12",
            "baseUrl": "https://github.com/obsproject/obs-deps/releases/download",
```

Nothing seemed to work, I tried changing the plugins to the ProgramFiles obs-studio plugins location as well but that didnt work as well.

[image]

<details>
<summary>heres the full logs</summary>

```log
09:53:28.055: CPU Name: Intel(R) Core(TM) i7-10750H CPU @ 2.60GHz
09:53:28.056: CPU Speed: 2592MHz
09:53:28.056: Physical Cores: 6, Logical Cores: 12
09:53:28.056: Physical Memory: 7979MB Total, 2096MB Free
09:53:28.056: Windows Version: 10.0 Build 26100 (release: 24H2; revision: 4351; 64-bit)
09:53:28.056: Running as administrator: false
09:53:28.056: Windows 10/11 Gaming Features:
09:53:28.056: 	Game DVR: On
09:53:28.056: 	Game Mode: Probably On (no reg key set)
09:53:28.057: Sec. Software Status:
09:53:28.058: 	Kaspersky Total Security: disabled (AV)
09:53:28.059: 	Microsoft Defender Antivirus: enabled (AV)
09:53:28.059: 	Kaspersky Total Security: disabled (FW)
09:53:28.059: 	Windows Firewall: disabled (FW)
09:53:28.063: Current Date/Time: 2025-07-06, 09:53:28
09:53:28.063: Browser Hardware Acceleration: true
09:53:28.063: Hide OBS windows from screen capture: false
09:53:28.063: Qt Version: 6.6.3 (runtime), 6.6.3 (compiled)
09:53:28.063: Portable mode: false
09:53:34.095: OBS 31.0.4 (64-bit, windows)
09:53:34.095: ---------------------------------
09:53:34.309: ---------------------------------
09:53:34.309: audio settings reset:
09:53:34.309: 	samples per sec: 48000
09:53:34.309: 	speakers:        2
09:53:34.309: 	max buffering:   960 milliseconds
09:53:34.309: 	buffering type:  dynamically increasing
09:53:34.321: ---------------------------------
09:53:34.321: Initializing D3D11...
09:53:34.321: Available Video Adapters: 
09:53:34.323: 	Adapter 0: NVIDIA GeForce GTX 1650 Ti
09:53:34.323: 	  Dedicated VRAM: 4126146560 (3.8 GiB)
09:53:34.323: 	  Shared VRAM:    4183293952 (3.9 GiB)
09:53:34.323: 	  PCI ID:         10de:1f95
09:53:34.323: 	  HAGS Status:    Enabled (Default: No, Driver status: Supported)
09:53:34.324: 	  Driver Version: 32.0.15.5597
09:53:34.324: 	  output 0:
09:53:34.324: 	    name=
09:53:34.324: 	    pos={0, 0}
09:53:34.324: 	    size={1920, 1080}
09:53:34.324: 	    attached=true
09:53:34.324: 	    refresh=144
09:53:34.324: 	    bits_per_color=8
09:53:34.324: 	    space=RGB_FULL_G22_NONE_P709
09:53:34.324: 	    primaries=[r=(0.594727, 0.361328), g=(0.345703, 0.554688), b=(0.157227, 0.106445), wp=(0.312500, 0.328125)]
09:53:34.324: 	    relative_gamut_area=[709=0.660716, P3=0.487061, 2020=0.349434]
09:53:34.324: 	    sdr_white_nits=80
09:53:34.324: 	    nit_range=[min=0.500000, max=270.000000, max_full_frame=270.000000]
09:53:34.324: 	    dpi=120 (125%)
09:53:34.324: 	    id=\\?\DISPLAY#NCP004D#4&22ecdc75&0&UID265988#{e6f07b5f-ee97-4a90-b076-33f57bf4eaa7}
[... 356 lines cut ...]
09:53:38.822:             - filter: 'User-defined shader' (shader_filter)
09:53:38.822:             - filter: 'Color Correction' (color_filter_v2)
09:53:38.822:             - filter: 'Apply LUT' (clut_filter)
09:53:38.822:             - filter: 'Apply LUT 2' (clut_filter)
09:53:38.822:             - filter: 'Apply LUT 3' (clut_filter)
09:53:38.822:             - filter: 'Apply LUT 4' (clut_filter)
09:53:38.822:             - filter: 'Apply LUT 5' (clut_filter)
09:53:38.822:             - filter: 'Apply LUT 6' (clut_filter)
09:53:38.822:     - source: 'StrezmerCam' (group)
09:53:38.822:         - source: 'pptLaptop' (ndi_source)
09:53:38.822:         - source: 'Main Cam' (dshow_input)
09:53:38.822:             - filter: 'Background Removal' (background_removal)
09:53:38.822:             - filter: 'User-defined shader' (shader_filter)
09:53:38.822:             - filter: 'Color Correction' (color_filter_v2)
09:53:38.822:             - filter: 'Apply LUT' (clut_filter)
09:53:38.822:             - filter: 'Apply LUT 2' (clut_filter)
09:53:38.822:             - filter: 'Apply LUT 3' (clut_filter)
09:53:38.822:             - filter: 'Apply LUT 4' (clut_filter)
09:53:38.822:             - filter: 'Apply LUT 5' (clut_filter)
09:53:38.822:             - filter: 'Apply LUT 6' (clut_filter)
09:53:38.822: ------------------------------------------------
09:53:39.327: YT: InitYTUserUrl() User is not signed
09:53:39.329: YT: InitYTUserUrl() User is not signed
09:53:45.130: [obs-backgroundremoval] Background filter activated
```

glikely, 2025-07-07:
There are two installers of the plugin for windows. One that works with OBS Studio <=31.0.4, and one that works with the 31.1.0 release candidate. Which one did you install?

glikely, 2025-07-07:
Wait, I can see the problem. You've got an older version of the plugin still installed. Delete QSerialPort.dll and obs-ptz.dll from the C:\Program Files\obs-studio\obs-plugins\64bit directory.

glikely, 2025-07-14:
Please do the following:
1. Delete all instance you can find of the obs-ptz and QSerialPort dlls from both Program Files and ProgramData.
2. Go into Add/Remove Programs and remove any instance of obs-ptz.
3. Run OBS and check the log to make sure it is no longer finding obs-ptz.
4. Reinstall PTZ Controls using the installer. **Do not** copy the files in manually.
5. Run OBS again and post the log here if you're still having problems

allanhanan, 2025-08-02:
I tried the following (TLDR: It didnt work)
1. Uninstalled obs-ptz from Add/Remove Programs
2. Deleted all instances of obs-ptz and QSerialPorts.dll from Program Files and ProgramData
3. Ran OBS to check logs and ensured no obs-ptz was found
<details>
<summary>heres the logs after that</summary>

```log
13:19:46.966: CPU Name: Intel(R) Core(TM) i7-10750H CPU @ 2.60GHz
13:19:46.966: CPU Speed: 2592MHz
13:19:46.966: Physical Cores: 6, Logical Cores: 12
13:19:46.966: Physical Memory: 7979MB Total, 2324MB Free
13:19:46.966: Windows Version: 10.0 Build 26100 (release: 24H2; revision: 4652; 64-bit)
13:19:46.966: Running as administrator: false
13:19:46.966: Windows 10/11 Gaming Features:
13:19:46.966: 	Game DVR: On
13:19:46.966: 	Game Mode: Probably On (no reg key set)
13:19:46.967: Sec. Software Status:
13:19:46.969: 	Kaspersky Total Security: disabled (AV)
13:19:46.969: 	Microsoft Defender Antivirus: enabled (AV)
13:19:46.970: 	Kaspersky Total Security: disabled (FW)
13:19:46.970: 	Windows Firewall: disabled (FW)
13:19:46.971: Current Date/Time: 2025-08-02, 13:19:46
13:19:46.971: Browser Hardware Acceleration: true
13:19:46.971: Hide OBS windows from screen capture: false
13:19:46.971: Qt Version: 6.6.3 (runtime), 6.6.3 (compiled)
13:19:46.971: Portable mode: false
13:19:52.714: OBS 31.0.4 (64-bit, windows)
13:19:52.714: ---------------------------------
13:19:53.750: ---------------------------------
13:19:53.750: audio settings reset:
13:19:53.750: 	samples per sec: 48000
13:19:53.750: 	speakers:        2
13:19:53.750: 	max buffering:   960 milliseconds
13:19:53.750: 	buffering type:  dynamically increasing
13:19:53.754: ---------------------------------
13:19:53.754: Initializing D3D11...
13:19:53.754: Available Video Adapters: 
13:19:53.757: 	Adapter 0: NVIDIA GeForce GTX 1650 Ti
13:19:53.757: 	  Dedicated VRAM: 4126146560 (3.8 GiB)
13:19:53.757: 	  Shared VRAM:    4183293952 (3.9 GiB)
13:19:53.757: 	  PCI ID:         10de:1f95
13:19:53.757: 	  HAGS Status:    Enabled (Default: No, Driver status: Supported)
13:19:53.757: 	  Driver Version: 32.0.15.5597
13:19:53.758: 	  output 0:
13:19:53.758: 	    name=
13:19:53.758: 	    pos={0, 0}
13:19:53.758: 	    size={1920, 1080}
13:19:53.758: 	    attached=true
13:19:53.758: 	    refresh=144
13:19:53.758: 	    bits_per_color=8
13:19:53.758: 	    space=RGB_FULL_G22_NONE_P709
13:19:53.758: 	    primaries=[r=(0.594727, 0.361328), g=(0.345703, 0.554688), b=(0.157227, 0.106445), wp=(0.312500, 0.328125)]
13:19:53.758: 	    relative_gamut_area=[709=0.660716, P3=0.487061, 2020=0.349434]
13:19:53.758: 	    sdr_white_nits=80
13:19:53.758: 	    nit_range=[min=0.500000, max=270.000000, max_full_frame=270.000000]
13:19:53.758: 	    dpi=120 (125%)
13:19:53.758: 	    id=\\?\DISPLAY#NCP004D#4&22ecdc75&0&UID265988#{e6f07b5f-ee97-4a90-b076-33f57bf4eaa7}
13:19:53.758: 	    alt_id=\\.\DISPLAY1
13:19:53.758: 	Adapter 1: Intel(R) UHD Graphics
13:19:53.758: 	  Dedicated VRAM: 134217728 (0.1 GiB)
13:19:53.758: 	  Shared VRAM:    4183293952 (3.9 GiB)
13:19:53.758: 	  PCI ID:         8086:9bc4
13:19:53.758: 	  HAGS Status:    Disabled (Default: No, Driver status: Unsupported)
13:19:53.758: 	  Driver Version: 26.20.100.7985
13:19:53.759: 	Adapter 2: Intel(R) UHD Graphics
13:19:53.759: 	  Dedicated VRAM: 134217728 (0.1 GiB)
13:19:53.759: 	  Shared VRAM:    4183293952 (3.9 GiB)
13:19:53.759: 	  PCI ID:         8086:9bc4
13:19:53.759: 	  HAGS Status:    Disabled (Default: No, Driver status: Unsupported)
[... 785 lines cut ...]
13:24:59.011:     - source: 'StrezmerCam' (group)
13:24:59.011:         - source: 'pptLaptop' (ndi_source)
13:24:59.011:         - source: 'Main Cam' (dshow_input)
13:24:59.011:             - filter: 'Background Removal' (background_removal)
13:24:59.011:             - filter: 'User-defined shader' (shader_filter)
13:24:59.011:             - filter: 'Color Correction' (color_filter_v2)
13:24:59.011:             - filter: 'Apply LUT' (clut_filter)
13:24:59.011:             - filter: 'Apply LUT 2' (clut_filter)
13:24:59.011:             - filter: 'Apply LUT 3' (clut_filter)
13:24:59.011:             - filter: 'Apply LUT 4' (clut_filter)
13:24:59.011:             - filter: 'Apply LUT 5' (clut_filter)
13:24:59.011:             - filter: 'Apply LUT 6' (clut_filter)
13:24:59.011: ------------------------------------------------
13:24:59.186: YT: InitYTUserUrl() User is not signed
13:24:59.186: YT: InitYTUserUrl() User is not signed
13:25:03.872: [obs-backgroundremoval] Background filter activated
13:25:43.462: adding 64 milliseconds of audio buffering, total audio buffering is now 64 milliseconds (source: Bg)
13:25:43.462: 
```
</details>
heres the dependency walker output

[image]

I also made sure the OBS bin and its DLL's are available in path

allanhanan, 2025-08-24:
I updated OBS to version 31.1.2 and reinstalled the plugin `obs-ptz-0.18.1-windows-x64-for-31.1.0-Installer.exe`, and it works

closed 2025-08-24 by allanhanan