glikely/obs-ptz #263: Translation - Danish

ETE-Design, 2025-07-11:
There seems to be a fault in your translation "Script" cause not all strings are translated - At least not all the Danish ones. Some are still in English though I have translated them and you have integrated them.

[image]

ETE-Design, 2026-05-10:
@glikely Did you fix this?

glikely, 2026-05-16:
No, haven't looked at it yet, but thanks for the reminder. I had forgotten.

closed 2026-09-20 by glikely

glikely, 2026-09-20:
I've made you a translation approver on Crowdin. I'm still trying to work out how best to use that too, but you should now be able to provide and approve the Danish translations without me having to look at it.