glikely/obs-ptz #273: PTZ Controls not showing in docks on OBS 31.1.2

iacjosiah, 2025-07-29:
I've installed the newest plugin that's compatible with 31.1.0, and even downgraded to OBS 31.1.0 but it's not showing up in the docks menu. I've tried downgrading the plugin and obs to previous versions as well and they're not showing up either.

wpeace3, 2025-07-31:
I'm seeing the same thing. The controllers have disappeared from OBS. 

I've reinstalled them, but no joy. 

Looking in the plugins directory, the files are not there. I've manually added them from my download directory, but the controls are still not showing in Docks.

BillyBob1354, 2025-07-31:
I nearly did this myself and almost missed that there are two versions for windows (zip or exe installer).
`obs-ptz-0.18.1-windows-x64`
and
`obs-ptz-0.18.1-windows-x64-for-31.1.0`

If the wrong one is used (zip or installer), it does not work and I get this error `Module 'C:\ProgramData\obs-studio/plugins/obs-ptz/bin/64bit/obs-ptz.dll' not loaded` (do you have this message in your logs?)

`obs-ptz-0.18.1-windows-x64-for-31.1.0.zip` works for me with OBS 31.1.2.
[image]

If not this, you could try what dev says in what seems to be a similar issue #262
(Here) https://github.com/glikely/obs-ptz/issues/262#issuecomment-3068743913