glikely/obs-ptz #281: obs-ptz will not load with OBS 32

lloydlavender, 2025-09-24:
obs-ptz will not load with OBS 32

AllenVM, 2025-09-27:
Yup, also having the same problem with the OBS 32.0.0 update.  Will be rolling back to OBS 31.1.1 until the problem is fixed - happy to test on Mac for you!

kq5m9, 2025-09-28:
Also issue on 32.0.1

kq5m9, 2025-10-02:
Same for Windows 10.

kq5m9, 2025-10-07:
From the OBS Project forum, I noticed the following info in a post regarding the 0.18.1 zip file: (https://obsproject.com/forum/threads/ptz-controls.143676/page-41)

> From Github you get:
> 
> - obs-ptz
>     - bin
>         - 64bit
>             - obs-ptz.dll
>             - obs-ptz.pdb
>             - Qt6SerialPort.dll
>     - data
>         - locale
>             - *.ini files
>  
> but it **MUST** be:
> 
> - data
>     - obs-plugins
>         - obs-ptz
>             - locale
>                 - *.ini files
> - obs-plugins
>     - 64bit
>         - obs-ptz.dll
>         - obs-ptz.pdb
>         - Qt6SerialPort.dll
>  

Upon making those changes, it worked for me. IHTH.

glikely, 2025-10-08:
> From the OBS Project forum, I noticed the following info in a post regarding the 0.18.1 zip file: (https://obsproject.com/forum/threads/ptz-controls.143676/page-41)
>
>
> Upon making those changes, it [...]

The zip layout is correct, It is organized for installing into the new plugin directory under c:\programdata, not the old location in Program Files

bigtreetech, 2025-10-19:
@glikely It seems as if it is not able to find the symbol for the frontend dock.

[2025-10-19 16-58-17.txt](https://github.com/user-attachments/files/22990821/2025-10-19.16-58-17.txt)

spartlow, 2025-11-04:
Yes, I deleted the obs-ptz.* files in program files plug-folder and resolved the problem. Ideally, the installer would do that (or instructions would be given to help people migrate), but my installations working now. Thanks.